eupn/bno055 #1: Function signature for usage code

AkashBhave, 2019-06-09:
First off, thanks for making this library.

I'm new to Rust, and don't have much experience with the `Result<>` return type. I used code under the "Usage" portion of the README (ending with `Ok(imu)`. 
However, my compiler says: 
```
the `?` operator can only be used in a function that returns `Result` or `Option`
```
I tried adding `-> Result<(), ()>` to the function signature, but I got different errors. What should the return type be?

My current code:
```rust
fn new() {
    let i2c_dev = I2cdev::new("/dev/i2c-1").unwrap();
    // Use the 0x28 address instead of 0x29
    let mut imu_dev = bno055::Bno055::new(i2c_dev, Delay).with_alternative_address();
    imu_dev.init()?;
    imu_dev.set_mode(bno055::BNO055OperationMode::NDOF)?;
}
```

hargoniX, 2019-08-14:
@AkashBhave Just in case you're still wondering, ? propagates the error (if there is one) further up the chain (meaning that it would return at this point with the error) as your function seems to be a function which is actually for an end program you should probably either .unwrap() or handle the error in another way here. You could still do it by changing the return type of your function to a result which contains the error type of this library but it's questionable wether you actually want to do that.

AkashBhave, 2019-08-14:
@hargoniX thanks for the help, I eventually got it cleared up by handling the error within the function.

closed 2019-08-15 by eupn