eupn/bno055 #4: Mint is not no_std

TheZoq2, 2019-11-16:
Hi, thanks for putting in the effort to making this lib!

It looks like you added `mint` as a dependency in 0.3.0, but unfortunately, it doesn't seem to compile without the standard library and as far as I can tell, there is no feature to enable/disable to fix that.

Am I missing something obvious here?

eupn, 2019-11-16:
Hey, thanks, I'll look into that! You can use the previous version that depends on `nalgebra` for now.

eupn, 2019-11-17:
I've made a PR https://github.com/kvark/mint/pull/47 that will make `mint` a `no_std` library. Until it's being merged, please stick to the previous version of the crate.

eupn, 2019-11-18:
PR is merged; I'll publish a new version of the crate as soon as `mint` will publish their's.

eupn, 2019-11-19:
Version is bumped https://crates.io/crates/mint/0.5.4. Updating the `bno055` crate.

eupn, 2019-11-19:
@TheZoq2 the crate should be able to build now after doing `cargo update`.

closed 2019-11-19 by eupn

TheZoq2, 2019-11-19:
Nice, thanks for the very quick fix!

PR #5 (2019-11-19): Fix mint version >= 0.5.4 and bump to 0.3.1
Closes #4.