eupn/bno055 #11: Example code building full example using delay

quietlychris, 2020-12-10:
Hello,

I'm working on getting an example of this up and running, and am a little confused about actually setting up the device using the embedded_hal traits, especially because it doesn't seem to be explicitly included in the `Cargo.toml` dependency list. 

Right now my example code looks like

```rust
use bno055::Bno055;
use linux_embedded_hal::{Delay, I2cdev};
use embedded_hal::blocking::delay::DelayMs;
use mint::Quaternion;

use std::error::Error;
use std::time::Duration;

fn main() {
    let mut dev = I2cdev::new("/dev/i2c-0").unwrap();
    let mut delay: dyn DelayMs = DelayMs {100u16};
    //    let mut delay = Delay {};
    let mut imu = bno055::Bno055::new(dev).with_alternative_address();
    imu.init(&mut delay).expect("Error initializing the IMU");
    imu.set_mode(bno055::BNO055OperationMode::NDOF, &mut delay)
        .expect("Error setting the IMU mode");

    loop {
        let quat: mint::Quaternion<f32> = imu.quaternion().expect("Error getting the IMU quaternion");
        println!("Quaternion: {:?}", quat);
    }

}
```
but am getting error relating to using the `Delay`. If I just use `let mut delay = Delay {};` it compiles, but then immediately panics with 
```
thread 'main' panicked at 'Error initializing the IMU: I2c(Io(Os { code: 6, kind: Other, message: "No such device or address" }))', src/main.rs:14:26
```

I'm pretty sure that my BNO055 is properly connected to the single-board computer (I'm using `$ sudo i2cdump 0 0x28` which seems to show that I do have bytes being read in from a device at that address), so my next debugging step is checking that the `delay` struct I'm passing to `init()` is properly constructed, but am a little confused about what that should look like. I see in the docs that it looks like it should be a `dyn DelayMs<u16>` but am not totally sure how to go about building that. Any advice would be really appreciated!

eupn, 2020-12-11:
Hello! `dyn Delay` means that we're using [dynamic dispatch](https://doc.rust-lang.org/1.8.0/book/trait-objects.html) to specify that we only care about types that implement the `DelayMs` trait. Usually, implementations of the `embedded-hal` have types that implement `DelayMs` via hardware-dependent means like ARM's `SysTick` or linux's `thread::sleep()` under the hood. For `linux_embedded_hal`, I think you've correctly imported the `Delay` struct. It should be initialized as a mutable binding to an empty struct, e.g.: `let mut delay = Delay {}`. Then, you should pass a mutable reference to this binding into `Bno055`'s `init()` function:

```rust
let mut delay = Delay {};
// ...
bno055
    .init(&mut delay) // <- here we're passing dyn DelayMs by mutable reference
    .expect("Error initializing the IMU");
```

For an address issue, you could try to remove a call to `with_alternative_address()` for it to probe a default address. And could you please share a `i2cdump`'s output that you're getting for `0x28` and `0x29` addresses?

Hope this will help!

eupn, 2020-12-11:
Added an `init` example into docs in #12.

quietlychris, 2020-12-11:
Ah, thank you! Knowing the right way to implement the Delay struct was helpful, since it helped to eliminate one of my unknowns, and I think I have it figured out. I'm not particularly used to the way that dynamic dispatch works, so thought the empty instantiation was just standing in for environment-specific code I had to implement. 

If it's helpful for anyone else, I've created a new repository with my (now working) code sample for the Odroid-C4 board at https://github.com/ucsb-coast-lab/read_bno055 , which also has a reference for the Adafruit-library-based Python implementation

Also, it might be worth noting that the most recent version of `linux-embedded-hal` on crates.io (0.4.0-alpha.0) doesn't play well with the existing code because of some missing trait implementations, so it's probably worth sticking with v0.3 for now.

closed 2020-12-11 by quietlychris