eupn/bno055 #15: std support

elaye, 2021-06-08:
Hello,

First of all, thank you for this crate. It's very useful for quickly getting started with the BNO055!

I use it on an embedded linux and there's a couple of features that I plan to develop for my needs by forking the repo.
For example: 
 - `std::error::Error` implementation for `Error`
 - `serde` serialization for the calibration data

However these features need the crate to support `std`, so I'll add it behind a feature flag.
My question is: since I'll develop this, will you be willing to consider merging this to your repo?
If so I'll cut the work in one branch per feature and submit merge requests.

eupn, 2021-06-08:
@elaye Hey, sure, go ahead! We should gate these behind an optional feature, so `no_std` still be the default mode as this crate is more targeted to `no_std` targets.

elaye, 2021-06-17:
Thanks for merging the changes!

closed 2021-06-17 by elaye