eupn/bno055 #27: embedded-hal 1.0.0-rc.3 and embedded-hal-async support

elpiel, 2023-12-27:
Hello,
I love that this crate exist and I would love to develop it further for the needs of the @AeroRust community.

I want to discuss how we could implement both rc candidates for `embedded-hal` and embedded-hal-async, along side the existing embedded-hal@0.2

elpiel, 2024-11-14:
@eupn what do you think about async support?

There are a few way of implementing this including e.g. `maybe_async` crate to write the code once but generate either blocking on async API, enabled by a feature.

The other option is to just duplicate the whole code and switch with `async`/`.await` and allow people to use both blocking (by default) and async (enabled with a feature) APIs.

I'd be very interested in getting this thing going as I'm working on some embassy-based flight controller implementation.
I can use a second core for the IMU blocking operations but I'd rather have async and not deal with blocking, especially for a crucial component of the control loop.

eupn, 2024-11-25:
@elpiel `maybe_async` sounds like a fine solution. Can't think of a scenario when someone would need to have both versions at the same time, I think it's okay to make them mutually exclusive.

elpiel, 2024-12-07:
I'll look into it and open a draft pr.

As for the exclusivity, even with it, I believe you can still add the dep. twice but with a different name and features.