eupn/bno055 #34: SemVer broken from 0.3.3 to 0.3.4 due to embedded-hal dependency update

quietlychris, 2024-02-18:
Hello, 

It appears that this library accidentally broke semantic versioning recently, which was the cause of a confused set of errors around trait bounds. In particular, `bno055 v0.3.3` used `embedded-hal v0.2.7`, but the update in cc365a9789ba0e41ea0dc3c61875c0b3b9e9c296 led to the `bno055 v0.3.4` release from a few week ago used `embedded-hal v1.0.0`. Those two libraries do not have compatible APIs around certain traits, which means that folks specifying `bno055 = "0.3"` (or basically anything besides `bno055 = "=0.3.3"` in their Cargo.toml file will likely see breaks. 

I would suggest cutting a new 0.4 release with the updated `embedded-hal v1` dependency, and yanking the `bno055 v0.3.4` release from crates.io. Additionally, it may be useful to look at https://github.com/rust-lang/rust-semverver for CI tooling or just as a precaution ahead of future releases.

Thanks for your consideration, and if you have any questions or there's anything I can do to support this, please let me know.

quietlychris, 2024-02-27:
@eupn Apologies for the direct ping, but just wanted to bring this to your attention in case you hadn't seen.

eupn, 2024-02-27:
@quietlychris sorry about that, f8a4151fbb4509406294cadc173d69831f33179c bumps the version to `v0.4.0`. The `v0.3.4` has been yanked.

quietlychris, 2024-03-02:
I appreciate it, thank you!

closed 2024-03-02 by quietlychris